heidelberg-makerspace/dai-makerspace-do-something #3: Paint the laser room

emka, 2016-06-04:
The room for a future laser cutter still needs to be painted.

- [ ] Move stored wood outside, (optional) find a better place for it
- [ ] Mop the walls
- [ ] Buy paint (white or blue/grey like the [https://www.epiloglaser.de/](Epilog) colors)
- [ ] Paint the walls
- [ ] Move wood back or find a better place

LukasFreeze, 2016-07-05:
Please wait with painting until I have carried out the necessary structural work - I'll keep you posted!

LukasFreeze, 2016-07-14:
wood can stay by the bookshelves, no time to move it back in and out again anyway (the laser will be [delivered](https://wiki.heidelberg-makerspace.de/wiki/Laser_Delivery) on August 1st!)

LukasFreeze, 2016-07-14:
the walls will be ready for painting after Wednesday, July 20th for sure. However, we should probably not go for the fancy "matching colors" option for various reasons including money and time consumption.

emka, 2016-07-26:
Painting might happen tomorrow.