heidelberg-makerspace/dai-makerspace-do-something #4: Label the light switches

emka, 2016-06-04:
It is currently impossible to tell which light belongs to which button at the entrance.

- [ ] Find proper names for all light locations
- [ ] Put labels next to the appropriate buttons at the entrance
- [ ] Print a 3D map with the names and lights (Lukas modeled the room already)

emka, 2016-06-14:
Names must be short (no space), suggestions: "Bar", "Entrance", "Sewing", "Maker Shelf".

LukasFreeze, 2016-06-14:
sounds good - would "Bar Front" still fit? Because one of the other switches controls the lights behind the bar (maybe "Photo Corner" or "Bar Back"?)
There is also one for the drinks room, could just be "Hands Off" imho ;-)

emka, 2016-06-15:
we could only identify the four switches for the main room, the others can be added easily once identified

LukasFreeze, 2016-06-20:
labeled the rest, 3D plan coming soon!

LukasFreeze, 2016-07-04:
done! 3D [top view of empty room](https://wiki.heidelberg-makerspace.de/wiki/File:Makerspace_empty_top_view.PNG) with lights drawn in is printed and on the wall above the switchbox :-)

closed 2016-07-04 by LukasFreeze