heidelberg-makerspace/dai-makerspace-do-something #6: Fix or replace drill press

LukasFreeze, 2016-06-05:
Problems:
- drill is rather old, chuck is +/- broken
- press only goes down 3cm ei

emka, 2016-08-16:
As it only goes down 3cm we probably need a replacement.

The best option for a new one in the lower price segment that I could find is [Bosch PBD 40](https://www.amazon.de/Bosch-Tischbohrmaschine-PBD-Parallelanschlag-Schnellspannklemmen/dp/B005OQEK9W), the cheaper ones from Einhell are probably too wobbly.

LukasFreeze, 2016-08-27:
how about we make this the next goal for the [Candy Shop](https://wiki.heidelberg-makerspace.de/wiki/Project:Candy_Shop) then? We can discuss the exact model once we get to around 200 €.

LukasFreeze, 2016-09-22:
We got a "new" one from a garage sale, just a few steps remain to close this issue:

- [x] see if it runs
- [ ] check if we have the right chuck key
- [ ] find a manual (Einhell TB13 / 5ECU)
- [ ] as always, write enough stuff for the wiki to make it accessible
- [ ] decide what to do with the old press (Dremel perhaps?)

emka, 2016-09-22:
Added a manual to the wiki page. Lukas, could you print the German (pp. 2-16) or English part?

I would sell / give away the old press as it occupies space and has no benefit over the new one.

LukasFreeze, 2016-09-22:
sure, I'll print it next week :+1:

LukasFreeze, 2016-09-27:
printed the relevant part of the manual today and will add it to the top bookshelf tomorrow!

LukasFreeze, 2016-09-29:
Both of our chuck keys are too small, I tried to measure as well as I could:
- shaft diameter should be 8 mm (maybe that makes it easy and there aren't too many different ones)
- total radius (incl. teeth) should taper from 12 to 10 mm
- teeth on the chuck are approx. 2 mm wide in the back and 3 mm wide in the back

Bauhaus only has the sizes we already have, but Entenmann probably carries more! If you **announce to buy one in this issue** you will get the money back from me.

LukasFreeze, 2016-10-05:
Entenmann does not have the chuck key size we need either, I will order it.

emka, 2016-10-05:
It turned out they are quite expensive, I will try to get the original one from the previous owner.

LukasFreeze, 2016-10-25:
Mounted the new drill press to the green workbench today, it can easily be removed (two screws) if the space is needed but is much more secure now. Once we have the chuck key(s), this can be closed!

LukasFreeze, 2016-11-02:
Chuck key arrived and is attached to the turnwheel of the press with a cable tie ring, not ideal as it could tighten over time but this way it shouldn't get lost! Feel free to improve it ;-)

Gave the old press to Klaus last week who is doing video projects involving milling

closed 2016-11-02 by LukasFreeze