heidelberg-makerspace/dai-makerspace-do-something #9: cleaning supplies order & accessibility

LukasFreeze, 2016-06-06:
We have some cleaning supplies but they are not seeing very much use!

- [x] dustpan and small brooms on the wall
- [ ] assign a color of rag to every purpose (e.g. blue for dishes etc, red for tables)
- [ ] rags on the wall (could use same system as for holding flyers + labels)
- [ ] buy some detergents & solvents (only have ethanol & acetone)

emka, 2016-08-10:
Acetone is almost empty as well

LukasFreeze, 2016-10-06:
I mounted three rags by the other cleaning supplies in the main room: a yellow one for tables, blue one for dishes and red one for "etc". The color is free to change whenever and we can easily hang more rags in other locations; the nails they hang on are labeled with the purpose.

Someone brought in dish soap a while ago (edited issue from "buy stuff" to "document stuff")...

emka, 2016-10-06:
I did not see the labels yet, we should make them bi-lingual to avoid confusion for non-English or beginner level speakers.

emka, 2016-10-08:
I created a short text on [Category:Cleaning](https://wiki.heidelberg-makerspace.de/wiki/Category:Cleaning), what else do you think is missing to consider this done?

LukasFreeze, 2016-10-08:
Adding more content on cleaning agents and maybe sending some reminders over the list if things are not left in an acceptable state will happen over time, I'll also add one or two little shelves to the cleaning corner at some point but mainly to make room elsewhere - the supplies should now be easily accessible!

closed 2016-10-08 by LukasFreeze