heidelberg-makerspace/dai-makerspace-do-something #12: take photos of hardware for the wiki

LukasFreeze, 2016-06-08:
Many tools, machines and materials have wiki pages already but the InfoBoxes are lacking a picture

emka, 2016-06-14:
List of pages currently missing pictures (checkmark means picture added):
- [ ] Dremel
- [ ] Screwdrivers
- [ ] Cordless Drill (Makita)
- [ ] Soldering Irons
- [ ] Disc Grinder
- [ ] Caliper
- [ ] Screwdriver Bits
- [ ] Twist Drill Bits
- [ ] Hole Saws
- [ ] Jigsaw
- [ ] Eccentric Sander
- [ ] Router
- [ ] Pyrography Burner
- [ ] Hex Keys

emka, 2016-07-20:
We took pictures of all above, still need to upload.

LukasFreeze, 2016-07-20:
to see pages that have InfoBoxes without pictures, check out these links:

- [tool pages without a picture](https://wiki.heidelberg-makerspace.de/wiki/Special:WhatLinksHere/File:Tool-default.png)
- [machine pages without a picture](https://wiki.heidelberg-makerspace.de/wiki/Special:WhatLinksHere/File:Machine-default.png)
- [material pages without a picture](https://wiki.heidelberg-makerspace.de/wiki/Special:WhatLinksHere/File:Material-default.png)
- and, maybe less relevant, [project pages without a picture]((https://wiki.heidelberg-makerspace.de/wiki/Special:WhatLinksHere/File:Project-default.png)

emka, 2016-07-21:
Old list is done, but Lukas linked many new pages (see above comment)

emka, 2016-07-27:
Took photos of all missing tools and machines that had an InfoBox. Materials are still missing, and I would ignore project pages as these are in responsibility of the project team.

emka, 2016-07-28:
All uploaded, tool pages are currently complete, the only missing machine page is the not yet delivered Laser Cutter.

emka, 2016-08-04:
Took more photos and added them to our wiki. Only two special materials are missing:
- [ ] Copper tape
- [ ] Fender washers

emka, 2016-09-21:
Took more photos today, unfortunately only in RAW format...

LukasFreeze, 2016-10-06:
I'm uploading about 15 new tool pictures, some don't have the _perfect_ depth of field but it was getting pretty late... I did not take too many material pictures this time.

emka, 2016-10-09:
Converted the RAW images and added them. I will add your links from [above](https://github.com/heidelberg-makerspace/do-something/issues/12#issuecomment-234102666) to the README as there will always be more.

emka, 2016-10-12:
This is a permanent issue as there will always be newly created wiki pages, added to the README.

closed 2016-10-12 by emka