heidelberg-makerspace/dai-makerspace-do-something #14: replace or fix "sign" on the outside

LukasFreeze, 2016-06-11:
Someone wrote "Makerspace" over the outside window with some kind of red tape, since many letters are missing by now it looks bad. If it is not fixed soon or the "artist" gives the tape to the Makerspace, we should take it down altogether and use, for example, the [flip dot display](https://wiki.heidelberg-makerspace.de/wiki/Project:Flip_Dot_Display) as an actual sign.

emka, 2016-06-14:
I took down the old sign, two letters were missing, the rest was hard to read and it seemed like we were just inactive.

emka, 2016-08-17:
I put a [printout of our logo with URL](https://github.com/heidelberg-makerspace/do-something/files/423003/Fenster.pdf) on the window that most people passing by were looking through as temporary solution until we come up with something more impressive.

closed 2016-08-17 by emka