heidelberg-makerspace/dai-makerspace-do-something #15: improve donation box visibility

LukasFreeze, 2016-06-11:
Lukas made a small sign reading 'donations' a while ago but it has since been "borrowed" at least once and is not visible enough, as some visitors suggested.

Ideas are, for instance:
- [ ] larger sign
- [ ] arrows

Pakue95, 2016-06-11:
What about a LED type sign, which blinks when a donation is made? I could contribute an Arduino Nano and a couple WS2818b. The sign would have to be battery (rechargeable) powered but that would be an easy task to implement. Depending on how soon we would need that sign a laser cut/engraved front plate out of acrylic glas would be a nice touch.

LukasFreeze, 2016-06-12:
sounds cool!

Pakue95, 2016-06-14:
took the initiative and started the sign today. Will involve lights and reward blinking when money is donated. Used some old display parts.

emka, 2016-06-15:
Pakue's sign looks cool already, I can't wait for it to shine in color! In the meantime I sticked the old one back on with double sided tape, sticks better than before.

emka, 2016-07-04:
New sign is up there, anything missing? Where is the battery?

Pakue95, 2016-07-08:
Everything is so far done and the battery is hidden behind the sign. Is a small 3000mAh battery pack by Amazon. I'm not sure how long it will last before it needs recharging but we'll see. 
An small addition could be a sensor to detect when someone donates something and show a reaction (blinking or similar). But this can be added later if someone is bored.

closed 2016-07-08 by emka