heidelberg-makerspace/dai-makerspace-do-something #18: label our equipment

LukasFreeze, 2016-06-13:
Lots of Makerspace tools could still use a "Makerspace" label and/or one that names the corresponding wiki page (have a look at the Makita drill as an example). Not everything can be labeled with the Labelmaker though, and unfortunately the labels currently don't stick...

- [ ] create "Labelmaker" wiki page
- [ ] find a way to use non-sticking portion of the current label cassette (i.e. a suitable glue to use it with)
- [ ] label tools/furniture/power cords that **definitely** belong to the Makerspace

emka, 2016-06-14:
The current label cassette actually works as intended, it is just difficult to remove the non-sticking part. Luzian is able to remove it.

emka, 2016-06-14:
I labeled a few things, it is difficult to tell what is missing.

emka, 2016-09-22:
@LukasFreeze, can we close this one?

LukasFreeze, 2016-09-22:
I suppose we can, it's too vague anyway :-P

Let's keep in mind the labelmaker exists whenever we find something "new"

closed 2016-09-22 by LukasFreeze