heidelberg-makerspace/dai-makerspace-do-something #20: bring old laptop battery (and other stuff) to recycling center

LukasFreeze, 2016-06-13:
There is an old laptop battery in the green metal drill box that should ideally be brought to a recycling center before it explodes. While you're at it, please empty the metal junk bucket and e.Waste as well ;-)

emka, 2016-08-02:
We have two bags of demolition rubble as well which are currently occupying storage space.

Pakue95, 2016-08-02:
[Here](http://www.heidelberg.de/hd,Lde/HD/Rathaus/Recyclinghoefe.html#standorte-und-regulre-ffnungszeiten) are some places around Heidelberg which can take the waste. The one in Kirchheim seems to be the easiest to get to. They are all free as long the amount is below one m<sup>3</sup>. 
We'll probably need a car for this.

LukasFreeze, 2016-08-02:
let's call it "renovation rubble", okay? ;-)

But yes, we *definitely need a car*

emka, 2016-08-04:
What about the door? Should it also go there?

LukasFreeze, 2016-08-04:
Yes, but if only a regular car is available we could sit on the door a little longer but get rid of the rest

emka, 2016-08-09:
Patrick and I brought everything except the door to the recycling center. The "non-recyclable renovation rubble" cost us a small fee. Done just in time before Sofienstraße will be closed completely.

closed 2016-08-09 by emka

LukasFreeze, 2016-08-10:
Thanks a lot guys & enjoy your Merit Mate!