heidelberg-makerspace/dai-makerspace-do-something #21: organize Mindstorms parts

LukasFreeze, 2016-06-16:
We bought cases for our Mindstorms kits, but have not found the perfect order yet - please think about what makes sense and organize the parts (additional separators coming soon, one of the cases already has enough)

- [ ] find a good system
- [ ] take a photo and add it to the [wiki page](https://wiki.heidelberg-makerspace.de/wiki/LEGO_Mindstorms)
- [ ] also add a description of the system to that page (e.g. "all wheels in compartment A, motors in...")

emka, 2016-08-02:
Lukas and Paul made some progress here

LukasFreeze, 2016-08-02:
Yes, albeit barely useful at the moment - we did not have time to label the compartments so the next users won't know what our idea was (photo coming up on the wiki anyway!)