heidelberg-makerspace/dai-makerspace-do-something #23: list library books on the wiki

LukasFreeze, 2016-06-27:
Besides our [Makerspace bookshelf](https://wiki.heidelberg-makerspace.de/wiki/Makerspace_Bookshelf), there are many interesting books and magazines two stories up in the DAI library! It would be great if there was a list of these titles on the ["Library" wiki page](https://wiki.heidelberg-makerspace.de/wiki/Library) - please transcribe the titles from the photo already on the page.

emka, 2016-08-17:
All books in the maker shelf have signature (Signatur) "MAK *", but I could not figure out how to search for that in the library online catalog. I asked Rebekah for advice how to run this kind of search.

LukasFreeze, 2016-08-27:
Mitja added the information on how to search the library catalogue to the wiki.

closed 2016-08-27 by LukasFreeze