heidelberg-makerspace/dai-makerspace-do-something #24: set up mail server

LukasFreeze, 2016-06-28:
We need a **real** discussion list, not just "reply to all" group eMails! The GNU tool [Mailman](https://www.gnu.org/software/mailman/) could be an option, but needs to be hosted on a server.

emka, 2016-06-28:
We will use our own installation of Mailman and Postfix. I am still learning the details of Postfix configuration. Setting up Mailman via [DebOps](http://debops.org/) might be an option.

drybjed, 2016-06-28:
Hello, DebOps author here. If you need any help with setting up Mailman, let me know. You can also check out the project's own [Mailman installation](https://lists.debops.org/) if you want.

emka, 2016-07-04:
Thanks for the offer! I was able to install mailman using DebOps, but there is still a lot of configuration I need to check. for example the default postfix configuration.

drybjed, 2016-07-04:
@emka Great. If you find something amiss, let me know. :-)

emka, 2016-08-20:
I am not able to receive mail from mailman yet, maybe postfix needs to be configured differently. In ```mail.log``` it says
```
warning: hash:/var/lib/mailman/data/virtual-mailman is unavailable. open database /var/lib/mailman/data/virtual-mailman.db: No such file or directory
warning: hash:/var/lib/mailman/data/virtual-mailman lookup error for "newsletter-bounces@lists.heidelberg-makerspace.de"
NOQUEUE: reject: RCPT from localhost.localdomain[127.0.0.1]: 451 4.3.0 <newsletter-bounces@lists.heidelberg-makerspace.de>: Temporary lookup failure;
```

The current configuration is now available at [debops-makerspace](https://github.com/heidelberg-makerspace/debops-makerspace).

Another open task is to enable Letsencrypt, I found docs on [ACME integration](https://docs.debops.org/en/latest/ansible/roles/ansible-pki/docs/acme-integration.html) but did not figure out how to configure anything yet.

drybjed, 2016-08-20:
@emka Keep in mind that currently DebOps does not deploy Postfix automatically, your host needs to be added to `[debops_service_postfix]` group for that to happen. For the server to accept incoming mail, you might want to enable at least `mx` and `network` Postfix capabilities in the role.

Were there any errors during Ansible playbook run? Also, what OS are you using for the host?

For ACME deployment, you need to setup the `nginx` server, with `debops.nginx` role. Since it's needed by the Mailman anyway, it's probably deployed already. For ACME deployment you need an additional PKI realm configured in the inventory, for example:
```
pki_realms:
  - name: '{{ ansible_fqdn }}'
    acme_default_subdomains: []
```
When you run the `debops.pki` role after `nginx` is configured, it should automatically try to get ACME certificates for the specified FQDN hostname.

If you still have any issues, come over to `#debops` channel on FreeNode, we will help you through. :-)

emka, 2016-08-21:
Thanks to the awsome support by @drybjed  on IRC, everything works fine now. The main problem was using latest Ubuntu, on Debian Jessie everything is set up as intended.

I will spend some more time to migrate our Mediawiki to the new server before we start using it.

emka, 2016-10-04:
We are still modifying mailman configuration. A Gmail account does not receive some mails or they are delayed for half a day, currently needs deeper investigation.

LukasFreeze, 2016-10-06:
I am trying a different account of mine today (zimbra), it works for self-subscribing to the newsletter and should also receive the sent mails, both of which are still not happening after three days of waiting for the Gmail account. Who else has a gmail account and would like to be a guinea pig for a while?

In the meantime, I'll set up the Lvl2 discussion list which may lead to the discovery of completely new bugs...

LukasFreeze, 2016-10-06:
- removed subscription fields from lvl2's HTML page since the list is exclusive and members are added by the organizers
- made archives private for the same reason (also turned off entirely during testing)
- adding members by "invite" is not ideal: the automatically sent eMail suggests several options to confirm subscription, replying does not work for all mail programs but visiting the confirmation link does
- posting to list takes 2-3 minutes but seems to work!

LukasFreeze, 2016-10-06:
there are two bugs that came to my attention during testing, there seems to be a general problem with embedded links:
- mailto-link in welcome eMail ends at the "-" (before "makerspace.de") and does not work of course - PC, aol.com
- confirmation link in "new subscriber invitation" mail causes "403/forbidden" error when clicked but works fine when copied - 2 PCs, zimbra and aol.com

drybjed, 2016-10-06:
The issue with link not working when clicked somewhere else is because the Mailman `nginx` configuration blocks untrusted referrers, only allowing links from the host's FQDN, domain, etc. You can change this in the `debops.mailman` configuration.

This is done so that malicious sites won't hijack the Mailman web interface and generate unsolicited mails using HTML forms.

emka, 2016-10-07:
I found something in the logs:
```
22102:Oct  7 11:01:42 lists postfix/smtp[22940]: 3sr3RT3HH0zDqGw: to=<example@gmail.com>, relay=gmail-smtp-in.l.google.com[2a00:1450:400c:c09::1a]:25, delay=0.73, delays=0.04/0/0.16/0.52, dsn=5.7.1, status=bounced (host gmail-smtp-in.l.google.com[2a00:1450:400c:c09::1a] said: 550-5.7.1 [2a01:4f8:c17:1e51::2] Our system has detected that this message does 550-5.7.1 not meet IPv6 sending guidelines regarding PTR records and 550-5.7.1 authentication. Please review 550-5.7.1  https://support.google.com/mail/?p=IPv6AuthError for more information 550 5.7.1 . v11si21441769wjr.95 - gsmtp (in reply to end of DATA command))
```

Our IPv6 Authentication & Identification is not yet configured correctly...

emka, 2016-10-07:
We were missing a reverse DNS entry for IPv6

LukasFreeze, 2016-10-07:
invitation sent to all unsubscribed lvl2 members, seems to work as intended now (even with Gmail and embedded links!)

emka, 2016-10-11:
This part is done, created a milestone for the rest.

closed 2016-10-11 by emka

emka, 2016-10-13:
@drybjed now I understood how nginx will block unknown referrers. This is causing trouble for everyone using webmail like Gmail because clicking the subscription confirmation will not work. What is the risk of disabling ```location_referers```, did you see mass subscriptions followed by spam in the past?
I would disable it by just setting
```
mailman__referers: '*'
```
is that the way to do it?

drybjed, 2016-10-13:
@emka No, I haven't seen any spammers. The idea behind restricting referers by default in DebOps was that you shouldn't become a spam source automatically after installing Mailman. Now that you know about this mechanism, you can certainly tweak/disable it.

I wouldn't suggest enabling referers for all domains, you can add Google, Outlook and any other domains that you deem fit to the `mailman__referers` list. Since it also holds the default domains, you should do it like this:
```
mailman__referers: '{{
  ([ mailman__site_domain ] +
   mailman__virtual_domains +
   [ mailman__fqdn, "*." + mailman__domain, mailman__domain ] +
   [ "google.com", "*.google.com", "outlook.com", "*.outlook.com") }}'
```
Most likely at some point I'll move the default referer domains to a separate variable so that adding custom ones will be easier.