heidelberg-makerspace/dai-makerspace-do-something #25: improve signs

LukasFreeze, 2016-06-28:
There are many sheets of paper around the Makerspace that contain important information but frankly look like crap. If you are feeling creative, please check with the original author (usually Lukas...) whether a certain sign is finished and if yes, feel free to create a better version - e.g. a laminated print, 3D print (rarely the best option), pyrography on wood etc.

Information may be rearranged and reworded but the original message must remain! Also feel free to add more signs if something important is missing.

Some examples of "finished" signs in need of improvement:
- [ ] Candy Shop sign (new one should go on the wall instead of the table - fewer chips there)
- [ ] "3 Simple Rules for 3D Printing" (just needs lamination)
- [ ] "Project Shelf" (needs lamination, but may be better in A3 instead A4?)
- [ ] guide to Cura/slicing

emka, 2016-10-12:
I put signs of our principles on the walls and pushed them to the [signs repository](https://github.com/heidelberg-makerspace/makerspace-signs/tree/master/principles).

LukasFreeze, 2017-02-20:
I designed a "word cloud" sign for the wiki URL, any thoughts? I want to engrave this on a piece of wood soon and would be happy about any feedback :-)

[image]