heidelberg-makerspace/dai-makerspace-do-something #28: fix Stepcraft

LukasFreeze, 2016-07-03:
the Stepcraft should finally become functional again! As far as I know the new electronics are almost ready to be mounted:

- [ ] find out which parts are still missing
- [ ] locate or re-order them
- [ ] finish soldering the board
- [ ] install & commission

emka, 2016-07-04:
I generated the part list in EAGLE: [Partlist.txt](https://github.com/heidelberg-makerspace/do-something/files/346554/Partlist.txt)

emka, 2016-07-04:
As far as I can see we are missing
* R31 (1k)
* R36 (4.7k)
* R38 (1k)
* X4 (mini USB B)

We have leftover
* 2x SMD-0805 2.7k
* 3x SMD TAN.2.2/20

Something does not add up here!

emka, 2016-07-06:
We identified and ordered all missing parts except for an optional backup battery.

emka, 2016-07-19:
All parts are soldered now. We are missing a PDI programmer for the ATXmega microcontroller. I ordered one which should arrive in two or three days.
In the meantime we should prepare the firmware to avoid another time-wasting surprise.

emka, 2016-07-25:
I borrowed a programmer and we were able to program the prebuilt firmware from the forum on Saturday. The baudrate was just half of what was configured, there might be a wrong clock setting.
We installed the PCB inside the Stepcraft, but were not able to connect any host software other than a serial console (tried Universal Gcode Sender and Chilipeppr).

Today my new PDI programmer arrived so we should be able to upload new firmware in the future.
We still need a new Mini-USB B cable as the old PCB had a (large) USB-B connector.

I was also able to compile the latest TinyG on Debian Jessie (```gcc-avr``` on Debian is old enough, it is too new for the TinyG codebase on current Ubuntu).

I am not sure yet which modifications have been done on the upstream TinyG source code by the Stepcraft TinyG PCB designer, we need to find that out.

emka, 2016-08-23:
Sebastian and Korbi programmed a bootloader two weeks ago, we did not make more progress on the firmware side so far.

emka, 2016-08-30:
We put the original control board  back in, using it works but still has the same forgetting Z height issue. Broke on milling head in the process. No progress on the new board yet.

emka, 2016-10-05:
We will try two approaches: populating a second PCB (the first one has a burning/evaporation mark, maybe traces got damaged) and USB sniffing the original control PCB.

LukasFreeze, 2016-10-25:
just stumbled upon something - are you aware of [this Stepcraft forum thread](https://www.stepcraft-systems.com/forum/fraesen/3619-nullpunkt-z-achse-ploetzlich-woanders) where a similar problem was caused by faulty grounding?

emka, 2016-10-26:
We did not attach the earthing cable yet, I will do that today

emka, 2016-11-02:
Still observing Z axis shifts, Andre suggested that the motor might be losing steps because the axis is not moving as easy as it should. WinPC-NC Starter does not support changing Z acceleration in the software, but there are two configuration files for our model which are just text files containing many variables and values.

emka, 2016-12-03:
I documented our new approach using an Arduino with grbl shield and 3x A4988 ("Keyestudio 3D CNC kit") as [Project:Grbl for Stepcraft](https://wiki.heidelberg-makerspace.de/wiki/Project:Grbl_for_Stepcraft)

emka, 2016-12-13:
Everything works fine now except for #105

closed 2016-12-13 by emka