heidelberg-makerspace/dai-makerspace-do-something #31: Create ToolInfoBoxes

emka, 2016-07-11:
I uploaded many pictures today, but the articles still need ToolInfoBoxes.

Check today's recent changes for a list, I will still upload a few more today.

LukasFreeze, 2016-07-13:
unresolved stubs Mitja is referring to (minus the ones I continued today):
- [ ] Coffee Machine (needs MachineInfoBox, but same general idea)
- [ ] Paper Cutter
- [ ] IFixIt Pro Tech Toolkit
- [ ] Material:Stone (needs MaterialInfoBox)
- [ ] Hot Glue Gun
- [ ] Spatulas
- [ ] Sensor Kit
- [ ] Scale
- [ ] Level
- [ ] Pincers
- [ ] Taps and Dies
- [ ] Hammers
- [ ] Material:Acrylic Glass (needs MaterialInfoBox)
- [ ] Material:Plywood (needs MaterialInfoBox)
- [ ] Knives
- [ ] Files

emka, 2016-07-19:
All of these pages now have InfoBoxes. Of course they could need more text.

closed 2016-07-19 by emka