heidelberg-makerspace/dai-makerspace-do-something #32: improve laser printer access

LukasFreeze, 2016-07-13:
Installing the [laser printer](https://wiki.heidelberg-makerspace.de/wiki/Laser_Printer) driver on various systems has proven rather problematic and we are currently only able to print witht the slicing laptop.
There are basically two options - **enable printing from the Mac Mini** or **set up a (Pi-based?) network compatible printer control module** to eliminate this problem.

emka, 2016-10-09:
For Ubuntu, there is [a bug report](https://bugs.launchpad.net/ubuntu/+source/splix/+bug/1102001). I will try to print from my laptop next time. If we get a running CUPS setup, we could use a Pi-based setup if still required, but maybe documenting how to print from any laptop is sufficient.

emka, 2016-10-12:
Works in Ubuntu, when adding the printer, select "Samsung", but NOT the recommended ML-1660. Choose model CLP-310 instead (foomatic/foo2qdpl) and select monocrome color at 600dpi.

LukasFreeze, 2016-10-25:
I put those lines into the wiki, further steps are outlined in #71 if we want to get into network printing anyway.

closed 2016-10-25 by LukasFreeze