heidelberg-makerspace/dai-makerspace-do-something #33: Create a table for the USB hub we'll connect the entry raspi to.

Pakue95, 2016-07-14:
Thanks to "Markuhn" we now got a fancy OPEN/CLOSED lever with API for our Makerspace. However it needs power for a Raspberry Pi so we ordered a 10 port 60W USB charger as power supply. This is certainly overkill for one Pi but the idea is to use it as a "charging center" for phones and similar.

- [ ] create a nice table/phone rest to support the charger and secure it from any inappropriate use.

emka, 2016-07-26:
There is a [project page](https://wiki.heidelberg-makerspace.de/wiki/Project:Powertable) in our wiki now.

Pakue95, 2016-08-02:
The charger is installed and is located at the entrance. An idea was to add some lighting in the two boxes of the table. However this is not an essential addition therefore I'll close this issue.

closed 2016-08-02 by Pakue95