heidelberg-makerspace/dai-makerspace-do-something #35: set up Amazon affiliate link

LukasFreeze, 2016-07-22:
nice Makers can already donate online shopping provisions to us via our [gynny page](http://www.gynny.de/projekte/2355), but Amazon is not supported there anymore. We could set up our own "partner" link and put it on the wiki pages where Amazon is a good sourcing option.

emka, 2016-07-23:
I looked into it recently. The affiliate link is connected to an Amazon account and can be created at https://partnernet.amazon.de/
I would prefer to have a separate account with something like vorstand@heidelberg-makerspace... email address, but we need to set up a proper mail server first.

LukasFreeze, 2016-07-23:
good point, just wanted to have the idea out there for whenever it makes sense :-)

LukasFreeze, 2016-08-10:
amazon.com has set up ["Amazon Smile"](http://smile.amazon.com/gp/chpf/about/ref=smi_aas_redirect?ie=UTF8&ref_=smi_ge_ul_lm_raas) a few years ago which allows users to select a charitable organization once and then automatically donates 0.5 % of nearly all purchases, but I haven't been able to find a German/European implementation yet...

emka, 2016-10-05:
We will set it up using ```orga@``` as email, depends on #73

emka, 2016-10-12:
Lukas, can you set that up as you are managing the financial side?