heidelberg-makerspace/dai-makerspace-do-something #37: improve Cura

LukasFreeze, 2016-07-22:
something for our Python meetup group (or whoever wants to try): Cura has ease of use and fast slicing speaking for it, but there are several shortcomings that might be fixable (plugins or altering the source?)

-[ ] **add '€' sign** after material cost - the last number in the calculations at the top, where duration and filament use are estimated, is the suggested donation, but few people get it as it lacks the unit
-[ ] **visualize other machine beds** like the Ultimaker's - would be nice to have at least the printer name in the back, Kossels also need a more complicated build volume than just a cylinder (Slic3r takes STLs as custom print volumes)

I know there was more but can't remember now, feel free to comment what you'd like to see!

LukasFreeze, 2016-07-24:
Cura-related code is found at the [Ultimaker repository](https://github.com/Ultimaker).

emka, 2016-10-05:
Dirk has made some progress here.

DirkToewe, 2016-10-06:
Yes indeed I did. I made a [GitHub project](https://github.com/DirkToewe/makerspace_cura_launcher) for it and tried to document it a little. You guys can just fork it and modify it. If you have any questions, best ask them here then we have them documented.

DirkToewe, 2016-10-06:
It should be enough to copy the [CuraLauncher.py](https://github.com/DirkToewe/makerspace_cura_launcher/blob/master/Makerspace%20Cura%20Launcher/src/makerspace_hd/cura_launcher/CuraLauncher.py) into the cura directory and execute it with python.

To add new/different printer bed models to the GUI, just add more `elif`-clauses to the if-statement in [line 77](https://github.com/DirkToewe/makerspace_cura_launcher/blob/master/Makerspace%20Cura%20Launcher/src/makerspace_hd/cura_launcher/CuraLauncher.py#L77-L78).

emka, 2016-10-18:
I put ```CuraLauncher.py``` in the slicing laptop's ```Cura_15.04.6``` directory and created a desktop link "Cura 15.04.6 - Makerspace customized". Launching it does not start Cura yet, even launching in CMD returns immediately.
Next step is to install an editor which can handle Unix line returns for further debugging.

emka, 2016-10-18:
There are two binaries bundled with Cura, ```pythonw.exe``` is called by the default shortcut, but that one does nothing when running ```CuraLauncher.py```. The other one, ```python.exe``` fails because ```distutils``` is missing.

emka, 2016-11-06:
Launching it works now, but I do not remember what @DirkToewe did to solve that.

We still need to create STLs for our printer models.

DirkToewe, 2016-11-06:
I blieve there was an unused `pkg_resources` import which i just removed. In addition to that i added a `~/Documents/Cura Printer Models` directory where the STL-files have to be inserted. Unfortunately, including custom printer model backgrounds does not yet work under Windows. I have not yet figured out why...