heidelberg-makerspace/dai-makerspace-do-something #38: document e.Bench

LukasFreeze, 2016-07-24:
we've had the e.Bench set up for a while now but very few things, especially components, are documented on the wiki.

-[ ] [oscilloscope](https://wiki.heidelberg-makerspace.de/wiki/?title=Oscilloscope&action=edit&redlink=1)
-[ ] simple [soldering irons](https://wiki.heidelberg-makerspace.de/wiki/Soldering_Irons#ZD-99_Soldering_Iron) (linked to sections that don't exist yet)
-[ ] small tools like breadboards and tweezers
-[ ] basic types of components and where they are found, e.g. Material:LEDs or Material:Resistors
-[ ] Material:Solder; flux and how to use it - could be nicely linked with the article demanded in issue #30

Pakue95, 2016-07-25:
I've got more free time starting from mid next month to document all of the equipment. Should be relatively little effort once the e.Bench is free (e.g. while do-something hours) and the respective manual is on hand.

LukasFreeze, 2016-12-13:
Whenever someone does start documenting the parts: I started an "[Electronic Compontents](https://wiki.heidelberg-makerspace.de/wiki/Category:Electronic_Components)" category in the wiki.

Pakue95, 2017-03-04:
Started inventorizing the electronic parts such as resistors an capacitors in the wiki. 
Feel free to add more stuff. 
https://wiki.heidelberg-makerspace.de/wiki/E.Bench_Inventory