heidelberg-makerspace/dai-makerspace-do-something #39: attach cloth to ceiling in Stepcraft room

emka, 2016-07-25:
There is sand coming off the ceiling in the Stepcraft room, which is really bad for the mechanics. To protect it, we can attach cloth to the ceiling at the lamp and staple it to the wooden walls.

Open questions:
- [ ] Do we have a suitable stapler?
- [ ] Which cloth can we use, does it need to be fire resistant?
- [ ] How to attach it to the ceiling?

When answered, get the cloth and attach it.

LukasFreeze, 2016-07-25:
The stapler on the pegboard should be fine, I used it for the Power Tool Hangar door as well. I think we should be able to use any cloth as long as it is fixed at the base of the lamp fixture and has no risk of touching the (anyways rather cold) tube itself.

emka, 2016-08-04:
White or similar might be best to reflect light downwards.

emka, 2016-11-20:
@YannickDupps will bring some fire-resistant black cloth. White would be much more expensive.

LukasFreeze, 2016-11-20:
In case he would have to buy that - we could use some of the Molton (black, fire resistant) fabric in the sewing corner! There is a very big piece :-)

YannickDupps, 2016-11-21:
I wouldn't have to buy it but if you already have black Molton at the Makerspace we could use that since it's the same thing as I wanted to bring.

LukasFreeze, 2016-11-21:
alright, then I'll see if we can get it done tomorrow during the sewing café!

LukasFreeze, 2016-11-22:
Done! Thanks @beanobeano and @YannickDupps for a lot of help.

We ended up just cutting the fabric to roughly the right size and trimming it after it was attached. We mounted a board under the lamp fixture to bear the main weight (the original mounting holes were terrible anyway, now there aremuch bigger screws and wall anchors in there)

closed 2016-11-22 by LukasFreeze

LukasFreeze, 2016-11-22:
Oh, one more thing: it's a bit darker now, but we can make a reflector if needed (not _too_ hard to take the lamp down again)