heidelberg-makerspace/dai-makerspace-do-something #40: Commisioning laser cutter

emka, 2016-08-04:
There are a few things left to do before we start cutting for everyone:

[ ] Buy Q-tips (for lens cleaning)
[ ] Buy CO2 fire extinguisher
[ ] Install and mount compressor
[ ] Find out name and where to get the air duct part for the back side (includes measuring dimensions)
[ ] Mount tubes closer to the ceiling (check laser door clearance)
[ ] Buy exhaust air tube
[ ] Build acrylic window with cutout and adapter for exhaust air tube
[ ] Install new window glass

I think it makes sense to create separate issues for the more difficult tasks and reference them here. Feel free to start with that.

emka, 2016-08-04:
We started a cut for #41 today. There were too many flames for my taste, before cutting anyting else we must mount the compressor. According to the manual the connector is on the back.

emka, 2016-08-09:
We got Q-tips and Patrick ordered a longer tube for the compressor.

LukasFreeze, 2016-08-10:
regarding the back exhausts: "Flachkanal" should be a good keyword, a quick search at Hornbach yielded [promising results](http://www.hornbach.de/shop/Umlenkstueck-Rotheigner-weiss-111x54-mm/8610773/artikel.html) but I don't have the right measurements of course ;-)
The cameo expert suggested foam fittings and a minimum distance to the rear wall of about 10 cm (main casing to wall, tubes can touch). We may have to alter or replace the metal tube fittings in the back to really save space, but that will be difficult!

emka, 2016-08-30:
We cut a piece of exhaust hood filter and put it above the lamellar filter to avoid damage by small pieces that were sucked in.

Pakue95, 2016-09-18:
The exhaust canals are finished as far as connecting to the laser is concerned. Only the connection from the door to the filter and filter to window have to be installed. Will likely be done by next week.

emka, 2016-09-20:
Did you finish the window part today?

LukasFreeze, 2016-09-20:
Yes, didn't get around to checking all issues tonight - sorry ;-)

closed 2016-09-20 by LukasFreeze