heidelberg-makerspace/dai-makerspace-do-something #41: Laser cutter donation box

emka, 2016-08-04:
We have separate donation boxes for several activities like sewing, 3d printing and workshop to allow designated funds. The laser cutter should get its own box, ideally manufactured using the laser cutter.

- [ ] Design a box.
- [ ] Assemle it.
- [ ] Mount it near the laser cutter (but accessible from outside when the door is closed).

emka, 2016-08-04:
Paul volunteered to design something and send us the files as he can not attend next Wednesday.

emka, 2016-08-04:
Paul designed one today and we cut it. Still needs assembly, a hinge and the metal locking part.

emka, 2016-08-10:
I glued the parts together. The design did not account for the thickness of of the material correctly and did not include a hinge, but the box is working for now. I 3d printed the part for a lock and added a hinge. We still need a lock, otherwise the box is operational.
[image]

emka, 2016-08-30:
Patrick brought a lock. When the box is locked, it can be bent on the bottom and all content can be removed. We need a better box design.

LukasFreeze, 2016-09-04:
I'll try to take care of it this week :-)

LukasFreeze, 2016-09-20:
@Pakue95 made and mounted one today, only needs to be added to the Wiki now!

closed 2016-09-20 by LukasFreeze