heidelberg-makerspace/dai-makerspace-do-something #43: Buy CO2 fire extinguisher

emka, 2016-08-04:
We need a 5-6kg CO2 fire extinguisher for #40, where can we get it? One option would be Ebay for approx. 70 EUR. It should have been tested recently to avoid additional cost or malfunction.

emka, 2016-08-04:
I will check out the store opposite of Bauhaus in Eppelheim.

emka, 2016-08-10:
The [shop I was talking about](http://www.fliesen-kamine-heidelberg.de/) turned out to be closed due to holiday until 29.8. but it is selling completely different products.

Bauhaus does not sell any CO2 extinguishers at all.
There might be a specialized shop in Weinheim, but I think we should just order on Ebay. @LukasFreeze what do you think?

There is one CO2 fire extinguisher in the kitchen, we could also try to find out which company the DAI is using for servicing.

LukasFreeze, 2016-08-11:
ebay sounds good, better to have this soon than to wait around for a possibly maybe okay-ish local option in this case :-P
As far as the DAI goes, we should team up with Mathias to push for better extinguishers in the whole building (powder /foam instead of water...)

emka, 2016-08-11:
Ordered a new 5kg one from a shop in Speyer

emka, 2016-08-23:
It arrived and is located at the laser room door, could be used on the e-bench or 3d printers as well.

closed 2016-08-23 by emka