heidelberg-makerspace/dai-makerspace-do-something #44: Install and mount compressor

emka, 2016-08-04:
Install and mount the designated compressor for the laser cutter. According to its manual the connector is on the back, might be hard to reach.

LukasFreeze, 2016-08-08:
a longer pneumatic tube will most likely be needed as well, do you know the diameter yet?

Pakue95, 2016-08-08:
The manual states 0,25" (6mm) at page 71. On Amazon a pneumatic tube this size (5m) costs about 11€ rated for 15bar (6x max pressure). If I ordered tonight it would arrive until Wednesday.

LukasFreeze, 2016-08-08:
sounds good! We can always shorten it later once we know where exactly the compressor can go. Make sure to keep the receipt as always ;-)

Pakue95, 2016-08-08:
Ok tube is ordered!

emka, 2016-08-15:
Tube is mounted, works much better with compressor turned on!

closed 2016-08-15 by emka