heidelberg-makerspace/dai-makerspace-do-something #45: Buy exhaust air tube

emka, 2016-08-04:
The filter for our laser cutter is currently pumping exhaust air on the laser cutter. We need a fitting exhaust air tube, find out the dimensions first and document them here and in the wiki.

lmnl99, 2016-08-10:
Ich habe die Teile auf der Hornbach Seite gefunden. Siehe Links

1: Übergangsstück  http://www.hornbach.de/shop/Umlenkstueck-Rotheigner-mit-Stutzen-weiss-111x54-mm/8610774/artikel.html

2: Flachkanal  http://www.hornbach.de/shop/Flachkanal-Rotheigner-weiss-111x54-mm-Laenge-1-00-m/8610771/artikel.html?varCat=S797##v8610770

3: Dickungsband  http://www.hornbach.de/shop/Dichtungsband-Rotheigner-15x5-mm-Laenge-2-m/8610879/artikel.html?sourceArt=8610771&url=8610771&trackArticleCrossType=cs

LukasFreeze, 2016-08-11:
Hier geht es um das Rohr nach draußen, oder? Der Flachkanal kommt hinten an den Laser und müsste ziemlich sicher größer als 11 cm im Durchmesser sein

emka, 2016-08-11:
Die von Luca verlinkten Teile sind für hinten am Lasercutter (#51), dafür gab es bis eben noch keine Issue. Der Schlauchdurchmesser dort ist kleiner als bei der Filterabluft, allerdings verjüngen die beiden hinteren Anschlüsse sich auch erst für den Schlauch.

LukasFreeze, 2016-09-18:
Es fehlen noch ein paar Teile (hole ich Montag), und die Leitung nach draußen muss gelegt werden, aber dann sind wir so weit - viel geschafft gestern!

LukasFreeze, 2016-09-20:
Fertig, alles zusammengebaut und dank @Pakue95 formschön nach draußen geleitet!

closed 2016-09-20 by LukasFreeze