heidelberg-makerspace/dai-makerspace-do-something #46: Create VisiCut settings

emka, 2016-08-04:
Create and push default VisiCut settings to Github and [send a pull request](https://github.com/t-oster/VisiCut/wiki/How-to-add-default-settings-for-your-lab)

emka, 2016-08-25:
Patrick was able to use the laser cutter with Visicut on a Macbook. We are able to cut from all major platforms!

Pakue95, 2016-09-18:
I've got a profile for our laser cutter running on my laptop which works nicely. Only the estimated time is far off as the Helix driver is used, which is a slower cutter than our Legend 36EXT. I haven't found the speed settings for the axis in the driver yet. For the non-Epilog laser cutters there is an option for this.
Also I'd like to wait for to laser cutter to work flawlessly before publishing the profile.

emka, 2016-10-05:
We could put the current profile in the repository anyway before everyone starts from scratch.

Pakue95, 2016-10-24:
Repository is up! See /heidelberg-makerspace/visicut-settings/

closed 2016-10-24 by Pakue95