heidelberg-makerspace/dai-makerspace-do-something #48: Design adapter to mount vacuum cleaner on circular saw

emka, 2016-08-09:
The circular saw has a trapezoidal dust exhaust, we need some part to fit our vacuum cleaner. Perfect for 3D printing!

Luzian designed the end that fits our vacuum cleaner in Onshape for another adapter in the past.

LukasFreeze, 2016-08-10:
This should be possible without drilling extra holes into the blade guard, one (M8?) is already there and the duct could reach _a bit_ into the trapezoid opening to keep it in place :-)

emka, 2016-10-05:
I started measuring and designing something. There is a hole through the saw, approximately 47mm deep. We have 3x40mm screws to screw into the adapter from the other side.

LukasFreeze, 2016-10-06:
We printed and tested it after you left, unfortunately it seems you designed the pin at the wrong spot. The trapezoid part fits well with the pin snapped off, but that pin (where the screw goes) is off by one diameter. Looking at the saw from "vacuum perspective" it should be further on the left, i.e. the left side of the pin is at the right side of the hole. You can have a look at it yourself, I put in into your box ;-)

The vacuum tube fits perfectly, hopefully not too well so the adapter does not come off with it. The hole for the 3 mm screw could be a tiny bit bigger, it holds but you can hear troublesome cracking noise...

emka, 2016-10-07:
You guessed correctly, I had a sign error calculating the pin position from one side. Uploaded an [updated design](https://www.thingiverse.com/thing:1812526) to Thingiverse and made the hole for the screw slightly larger.

emka, 2016-10-18:
Measuring the position of the pin is really tricky, it still did not fit.

emka, 2016-10-19:
It fits now, unfortunately the pin breaks off easily. We reinforced it with glue and a short threaded rod and put duct tape at the edges.

closed 2016-10-19 by emka