heidelberg-makerspace/dai-makerspace-do-something #49: Clamping holes in the large workbench

emka, 2016-08-09:
We need clamping holes in one side of the large workbench. While you are at it, we also need some nuts in the workbench to mount the scroll saw.

Please make sure to talk to Lukas about the positioning and sizes first.

LukasFreeze, 2016-08-10:
Once someone is assigned to this issue I'd suggest we buy a couple more clamps and make sure they are all compatible

LukasFreeze, 2016-11-21:
The common size for bench dogs ("Bankhaken") and benchtop clamps seems to be 20 mm.
Here's an [example of good quality parts](https://www.amazon.de/Veritas%C2%AE-Messing-Bankhaken-Paar-L%C3%A4nge-60/dp/B00149DLHA/) (check the "frequently bought together" section) - pretty expensive if we want something nice! We may be able to make lots of compatible clamps ourselves though, and these commercial solutions don't have to be more than inspiration ;-)

LukasFreeze, 2016-11-21:
Personally, I am against drilling any holes before the workbench finally gets its wiki page btw :-P