heidelberg-makerspace/dai-makerspace-do-something #51: Air duct part behind laser cutter

emka, 2016-08-11:
The current tube mount occupies too much space behind the laser cutter. We need a flat part with identical fitting to save some space and be able to move the laser cutter closer to the wall.

emka, 2016-08-17:
We have some [image] now. The biggest one I could find has a [12.5mm diameter](http://www.hornbach.de/shop/Flachkanal-Umlenkstueck-Rotheigner-weiss-220x54-mm-mit-Stutzen/8610784/artikel.html).

Lukas took a picture of a custom construction in another makerspace.

emka, 2016-08-17:
[image]
[image]

LukasFreeze, 2016-09-07:
André and I measured the hose today and "d" in the above drawing seems to be wrong - we got 10 cm (without rubber seal), could that be? Maybe this means we'll find a readily available solution!

emka, 2016-09-08:
No, there is an additional piece of metal attached on the one I have drawn. It reduces the diameter from "d" to 10cm. Maybe we could start with 10cm anyway instead of reducing it later, but that might cause some particles being stuck between the larger hole on the laser cutter and our custom plate. Another question for support ;)

LukasFreeze, 2016-09-15:
Although the installing technician said 10 cm room in the back were enough, today I was told to leave 20 - 30 cm to prevent overheating. This may eliminate the need for a flatter solution but we can still try to solve this issue for aesthetic reasons and as an experiment...

LukasFreeze, 2016-09-16:
Current distance to the wall is 26 cm, so we don't really have room for improvement according to the tech.

closed 2016-09-16 by LukasFreeze