heidelberg-makerspace/dai-makerspace-do-something #53: Laser stops firing after extended use

Pakue95, 2016-08-24:
TL;DR – Laser stops without notice on longer runtimes - likely cooling issue.

I noticed today that after extended use of the laser cutter (ca. 30min of runtime spread out over ~60-80min and high power settings) the laser stops working but continues to follow its job. 
After inspecting the cut piece a reduction of power can be seen before the laser went out.
There is no error message on the display nor does the cutter stop. After that incident I let the machine cool down for 10-15min and restarted the job, but the failure occurred after a few minutes of cutting again. All in all it happened 3-4 times until i was able to finish (kind of) the cutting.

It seems to be some kind of cooling issue as the runtime was longer between the failures the longer the laser was off. We should talk to one of the Epilog guys to see if this is a known problem and if it is fixable. It's really odd as these cutters are usually used 24/7 in other workshops.

- [ ] Search whether this is a known problem with this laser cutter
- [ ] Talk to Epilog 
- [ ] get it fixed

Pakue95, 2016-08-24:
Jop, seems to be a [known](http://www.sawmillcreek.org/showthread.php?168626-Help!-Epilog-laser-stops-firing-halfwa-y-through-job) issue. The guys here say it might be a clogged fan (could be as it is a used laser cutter) and the person with the problem got a new laser tube from Epilog after consulting the support.

LukasFreeze, 2016-09-07:
should I contact support then? We'll need to have #51 closed by the time they come so the laser is not moved more often than necessary

Pakue95, 2016-09-08:
I agree. Maybe we should test the failure issue once more when the cutter is completely "cooled down" by cutting a piece of wood/acrylic until the laser cuts out/decreases in power. We could show the piece then to the support and know exactly how long it takes to shutdown.

emka, 2016-09-08:
Pakue95's idea sounds good. I think we should contact support as soon as possible to avoid the impression that it might have worked fine before.

LukasFreeze, 2016-09-15:
I reached out to support today, will keep you posted about any progress!

LukasFreeze, 2016-09-15:
Okay, just got off the phobe with a technician. Before anyone will come and check or even replace the tube, we'll have to make sure it is not warm air in the room causing our problems. We'll need to:

- [ ] lead exhaust air outside
- [ ] measure room temperature (28 °C or higher are known to cause performance drops)
- [ ] make sure there are 20 - 30 cm of room behind the laser. This should be the case, but may prevent us from further pursuing #51 ...

Pakue95, 2016-09-18:
The exhaust air is now redirected outwards but without any difference in runtime (12min) of the laser. Further troubleshooting is up to the Epilog technicians as we've done everything to enable best possible cooling of the laser.

LukasFreeze, 2016-09-20:
cameo is aware of our problems and will contact me when a tech is available...

LukasFreeze, 2016-09-26:
I reached out to support again as nothing seems to have happened so far. Will keep you posted!

emka, 2016-10-05:
New status: we will get a service date in the next two weeks

LukasFreeze, 2016-10-12:
Service tech will come next Thursday, Oct 20th!

We need to run another test at 50 % power today to rule out problems with the power supply.

Pakue95, 2016-10-13:
Test with 50% Power resulted in 28min runtime.

LukasFreeze, 2016-10-20:
The technician found a rather surprising and luckily simple fault: the board controlling the cooling fans in a "smart" Silent Mode only worked if switched to "ON" manually. Replaced the board, test run at 100 % power was terminated after there was still no problem after 20 minutes!

closed 2016-10-20 by LukasFreeze