heidelberg-makerspace/dai-makerspace-do-something #54: Dampen workshop noise

emka, 2016-08-29:
The workshop noise is affecting all other rooms. Especially when there is a presentation in the main room or when louder power tools or hammers are in use, this disturbs all other people.

Can we build some noise dampening measures, "noise suckers" as on the walls in sound studios, and fire-proof noise-blocking curtains towards the main room?

LukasFreeze, 2016-09-05:
Not sure how practical "noise suckers" would be as lots of wall space is or will be used for storage.

I like the general idea though, do you think those heavy transparent strips they use in storage rooms could work? When Ciaran and I drafted the current layout we always thought it would be a pity to hide the workshop behind a wall or doors as it is nice to see some activity (albeit sometimes noisy)...

During (announced) presentations, I say we should also just prohibit loud work in general, like we do when there are DAI events upstairs.

emka, 2016-10-05:
In an exhibition of modern technology I found an inflatable noise dampening wall, "[CENO Wall](https://www.sattler-global.com/textile-architektur/laermschutz-1758.jsp)" (product launch 2016), maybe that is suitable as temporary noise blocking wall.

LukasFreeze, 2016-11-17:
An architect friend of mine suggested hanging some fabric from the ceiling and on exposed walls to greatly reduce the "funnel" effect the workshop currently has. I'll screen the fabric selection during next week's sewing café to see if we can try that

LukasFreeze, 2017-02-20:
A suggestion that came through the [Verbund offener Werkstätten](http://www.offene-werkstaetten.org/) mailing list this weekend: fabric-covered frames filled with stone wool!

I think this could be pretty cheap and effective, any opinions?