heidelberg-makerspace/dai-makerspace-do-something #57: Prusa B x-offset

Andi-Heidelberg, 2016-09-07:
Some offset in printed part occured while printing today, We. 07.sep .

emka, 2016-09-07:
MIght be related to #52 but the cause is not clear yet.

LukasFreeze, 2016-09-07:
Not sure if related, but the x-axis belt is way too loose and needs to be tightened in the course of this repair.

emka, 2016-09-21:
The first print worked fine today, second is still running.

LukasFreeze, 2016-09-26:
If prints work out again this Wednesday we could close this for now, but the loose belt remains an imperfection that should be tackled the next time parts on the machine need to be altered anyway.

This time, reducing the "X-Y-jerk distance" (distance allowed to travel at full speed rather than gradual acceleration) from 15 to 5 mm was all I did.

beanobeano, 2016-10-04:
Did a hi-tech hack to tighten the belt with a piece of bent coat hanger.
Running a test print now.

LukasFreeze, 2016-10-05:
two successful and long runs later, I'd say this is closed but feel free to improve @beanobeano's hack (e.g. adjustable with screw)

closed 2016-10-05 by LukasFreeze

Andi-Heidelberg, 2016-10-06:
This was simple and clever, and it works! Great! :-)

<div>-------- Ursprüngliche Nachricht --------</div><div>Von: beanobeano <notifications@github.com> </div><div>Datum:04.10.2016  13:57  (GMT+01:00) </div><div>An: heidelberg-makerspace/do-something <do-something@noreply.github.com> </div><div>Cc: Andi-Heidelberg <dekonstruktor@gmx.de>, Author <author@noreply.github.com> </div><div>Betreff: Re: [heidelberg-makerspace/do-something] Prusa B x-offset (#57) </div><div>
</div>Did a hi-tech hack to tighten the belt with a piece of bent coat hanger.
Running a test print now.