heidelberg-makerspace/dai-makerspace-do-something #58: Kossel A probe does not always trigger

LukasFreeze, 2016-09-08:
The Kossels have been "expert only" for long enough, it's time to finally fix them! Andrea and Johannes are working on Kossel B at the moment but all Kossel A might need is a more stable piece of shrink tube (or a small 3D printed part) to ensure the Z probe triggers every time.

LukasFreeze, 2016-09-18:
Luzian replaced the faulty part with an all-metal version and found the values to [re-calibrate the printer](https://wiki.heidelberg-makerspace.de/wiki/?title=Kossel_Calibration),  but Arduino suddenly kept throwing "compiler errors" without any further details. @Pakue95 guessed it may be outdated libraries, stepping down from Arduino 1.6.0 to 1.5.7 did not help...

emka, 2016-09-18:
I can look into it if you show me the errors.

emka, 2016-10-05:
The older version failed even with an empty sketch, showing the compilation call ```g++ <...>``` as error. Re-installed the current Arduino IDE. ```stdio.h``` introduced a ```typedef fpos_t``` with the same name as a ```struct fpos_t``` in the firmware, renaming the struct in the whole firmware codebase to ```marlin_fpos_t``` fixed this copilation issue.

Luzian can continue here.

Titan-OX, 2016-10-19:
fixed Arduino problem for this printer, updated calibration values and replaced the Z-probe spring with a stronger one.

next step: test print!

LukasFreeze, 2016-10-27:
Z-probe deploy values are off by a few mm, back to calibration...

Titan-OX, 2016-12-01:
The auto bed leveling dosen´t works in the Y axis. Lukas supposed a sign error.

LukasFreeze, 2016-12-05:
I was talking about the variable "Z_PROBE_OFFSET" but the values make sense to me and a sign error would cause a 10 mm difference, not just the tiny bit we're seeing. Remeasuring an correcting the Y offset by .2 mm or so could make the difference...

Here's how to do it (writing more instructions right now): [Kossel Calibration](https://wiki.heidelberg-makerspace.de/wiki/Kossel_Calibration)

LukasFreeze, 2016-12-15:
Bed leveling is still weird (larger nozzle-to-bed gap towards the back) but I noticed the secondary fan could crash into the probe retractor and updated the settings (see wiki). Measuring the offsets with a caliper rather than the machine itself did yield different values but no visible improvement.

I'm clueless on how to this right from here, would start using a "guess" approach and alternate between larger and smaller y offsets to see if _anything_ changes...

LukasFreeze, 2016-12-15:
Nope, it doesn't seem to matter what you enter for the Y offset (-3 worked just as bad as -7.2 or -13)...