heidelberg-makerspace/dai-makerspace-do-something #59: Ceiling mount for projector

emka, 2016-09-11:
Build a lockable ceiling mount for the projector. Mount it to the ceiling in the optimal distance for our screen.

LukasFreeze, 2016-09-16:
--- HOLD UNTIL FURTHER NOTICE ---
Apparently, the roller blind we are using as a screen has to be mounted further from the windows for safety reasons. The projector can not be mounted anywhere until this is clear (only heard about it today by chance...)

LukasFreeze, 2016-09-20:
We'll have to move the screen forward a bit, then we can see where to mount the projector. Hopefully it will still be possible without reducing the image size...

LukasFreeze, 2016-10-06:
Took the screen down today, hopefully able to re-mount it next week. Whoever had mounted it first did a terrible job, we'll have a more stable solution soon...

LukasFreeze, 2016-10-15:
The screen is re-mounted about a meter further into the room but there should still be spots to place the projector for a good image size. Went for a location slightly to the right of the room so the emergency light and power supply chain in the center won't get in the way.

LukasFreeze, 2016-10-25:
I have access to ladders and @Pakue95 will bring in a ball joint from an old camera mount soon, so we are nearly ready to get started on this!

Pakue95, 2016-11-11:
The mount for the projector is finished so far. We just need to drill holes in the ceiling and mount a piece of wood to screw the quick release into.

Pakue95, 2016-11-17:
Beamer sits and fits on the ceiling. HDMI and power cables are installed.
We should consider removing the boxes on top of the rollable shelf and convert it to the presentation/video laptop spot. Some general adapters (VGA, USB-C, miniDisplayPort etc.) to HDMI would also be a nice addition. Nevertheless this issue is done. *yay*

closed 2016-11-17 by Pakue95

LukasFreeze, 2016-11-17:
The file cabinet is a better option for the "laptop station" as the presenter can stand there during a talk. We could also store the adapters you mentioned in that cabinet (if we have money or the parts themselves donated...) ;-)