heidelberg-makerspace/dai-makerspace-do-something #60: Establish Embroidery

GesaHillmer, 2016-09-13:
The [Pfaff creative 1.5](https://wiki.heidelberg-makerspace.de/wiki/Pfaff_creative_1.5) sewing machine is able to embroider as well. With the corresponding software - [Premier+ Embroidery Intro](https://wiki.heidelberg-makerspace.de/wiki/Premier%2B_Embroidery_Intro) - you can have a look at designs already available or design your own.

Steps to get the embroidering going:

- [ ] download & install the software
- [ ] get familiar with the software: have a look at the designs
- [ ] get familiar with the software: change the designs
- [ ] make own designs

LukasFreeze, 2016-09-14:
...and everything is documented on the corresponding wiki pages as well, of course ;-)

chombium, 2016-09-14:
Cool... thanks for the hard work. When are we going to embroider the first T-Shirts? :)

I've gone briefly through the Pfaff creative 1.5 manual and I saw that the connection is via USB. Maybe we can reverse engineer the communication protocol and use some other tools and OSs to connect to the machine or maybe RPi or even arduino :)

Here are few links for USB sniffing on M$ Windowz (I haven tried my self)
http://desowin.org/usbpcap/tour.html
http://desowin.org/usbpcap/

And of course the user manual for the machine. I'm sure you have it already, but I'm adding it here for thread consistency:
http://www.pfaff.com/SiteMedia/PFAFF/Products/Machines/creative-line/creative1_5/Manual/creative_1-5_manual_EN.pdf

BR,
Jovan
P.S. Unfortunately the Sewing Cafe on Tuesdays is too early for me :(

emka, 2016-09-14:
I am also interested on reverse engineering the Stepcraft USB controller commands. Unfortunately the laptops we were using with Stepcraft software are already running Windows 10, I am not sure if usbpcap will work. Another option I read about is using a Unix host running Windows in Virtualbox and capturing USB with wireshark on the host.

Pakue95, 2016-09-15:
Can we just use another USB sniffer for the task? Something like [this](http://freeusbanalyzer.com)?

chombium, 2016-09-15:
@Pakue95 sure. One way or another we need to get the commands :)

emka, 2016-09-15:
Yes we could use any COM port sniffer, I found a few other options:
* Microsoft [Portmon](https://technet.microsoft.com/en-us/sysinternals/bb896644.aspx), use the filter rule "*write;*read"
* Eltima [Serial Port Monitor](http://www.eltima.com/de/products/serial-port-monitor/), 14 days demo version
* hub4com, part of [com0com](http://com0com.sourceforge.net/)

GesaHillmer, 2016-09-18:
The upper thread kept breaking during the first attempt of embroidering something although I used embroidery thread. I tried re-threading the machine and reducing the tension but that didn't help. I also cleaned the lower part of the machine around the bobbin-area.
As next steps it would probably be good to try different thread (just in case) and to reduce the speed.

GesaHillmer, 2016-10-04:
The next time I used the embroidery machine it was working fine. I put the spool pin in the vertical position before threading instead of horizontal. 
To get good results the embroidery fleece and fabric need to be taut in the frame; otherwise the bobbin will be raised.
I started showing people in the Sewing Café how to use the embroidery function. Next will be interested people during the Wednesday Making Hours.

LukasFreeze, 2016-10-04:
Awesome! I created a new [embroidery category](https://wiki.heidelberg-makerspace.de/wiki/Category:Embroidery) in the wiki, linked all the pages that are relevant so far and linked it from the table on the main page :-)

slpfrsly4, 2025-04-25:
Establish Embroidery is the art of decorating fabric or other materials using a needle to apply thread or yarn, and it has evolved into both a traditional craft and a modern form of creative expression. From hand-stitched patterns passed down through generations to high-tech machine embroidery used [embroidery fresno clovis](https://www.embroideryfresno.org/) in fashion and branding, this art form allows for limitless personalization and storytelling.