heidelberg-makerspace/dai-makerspace-do-something #61: 3D Printer emergency switches

JohannesGruhler, 2016-09-14:
we need emergency switches for the Prusa i3 printers.

- [ ] 3D print cases
- [ ] test switches Ciaran donated
- [ ] install (cutting the 12 V supply, not mains voltage)

emka, 2016-09-22:
I think they are done, @JohannesGruhler is there anything left to do?

JohannesGruhler, 2016-09-25:
perhaps emergency switch for Kossel B, but first we need to make issue #62 (Kossel B rewiring).

LukasFreeze, 2016-09-26:
As far as I remember the new Kossel power supplies should have switches on them, we could just engrave an arrow or similar into the new cases (see #62 for progress) and not buy any further emergency buttons.

@JohannesGruhler could now add the switches to the lists of modifications on the Prusas' wiki pages and this could be closed :-)

closed 2016-09-30 by JohannesGruhler