heidelberg-makerspace/dai-makerspace-do-something #62: Kossel B rewiring

LukasFreeze, 2016-09-14:
Andrea is redoing the wires of the Kossel B as it could use a better power supply and the original wires are too short in some places.

- [ ] feet for the whole printer - @emka is 3D printing the bases, we'll use M6 x 50 (or 60) mm hex-head screws that screw directly into the aluminium extrusions
- [ ] case for power supply - @emka is cutting the wooden parts
- [ ] wiring once power supply is re-established

emka, 2016-09-15:
I gave the feet design to Andrea, she will print them and prepare the screws.

LukasFreeze, 2016-09-22:
Andrea attached the feet and an HDF baseplate (cut by Johannes with the scroll saw) for the board to sit on. Rewiring will hopefully happen next week.

emka, 2016-09-22:
I cut parts for the case with the jigsaw. Some dimensions were a bit off and we will redo it with thinner wood on the table saw.

LukasFreeze, 2016-09-29:
rewiring should be finished next week!

emka, 2016-10-12:
Andrea finished the rewiring and printed a calibration cube. She took the wooden parts home to glue them, probably done tomorrow.

emka, 2016-10-20:
All done. There are just a few optional improvements remaining like a red circle around the power button and smoothed out corners.

LukasFreeze, 2016-10-27:
one more thing, sorry: added full metal Z probe and calibrated today, test print pending

JohannesGruhler, 2017-04-20:
Kossel B update 
- I removed the objekt fan because of (the fan was to big and to strong the nossel had problems to heat up)
- I replaced the Polyimide Tape
- I oiled the autoleveling construction with WD-40 (problems with the trigger)

LukasFreeze, 2017-04-20:
...and, most importantly, a firmware re-flash got the bed leveling to work again!

@JohannesGruhler, how did your test print turn out? You can close this issue if you were satisfied ;-)

JohannesGruhler, 2017-05-03:
There are new problem's with the extruder and I am designing new parts for it.

JohannesGruhler, 2017-05-20:
with the new extruder kossel B prints good results

closed 2017-05-20 by JohannesGruhler