heidelberg-makerspace/dai-makerspace-do-something #63: compressor for air assist runs hot and pops its tube

LukasFreeze, 2016-09-15:
The compressor assisting the laser cutter is malfunctioning because it is a membrane pump and the factory default of the laser's inlet valve is for a tank pump (Kesselkompressor).

Our cameo technician is walking me through flipping that valve on 2016-9-19 at 2 pm on the phone (feel free to come by), it'll be documented like any other change to a machine once I know how to do it.

Pakue95, 2016-09-18:
Also the pneumatic coupling is also quite damaged. I've already ordered new ones however the supplier sent the wrong ones… Maybe next week the right ones arrive.

LukasFreeze, 2016-09-19:
It took a while due to spatial restraints in the laser room and even tighter corners in the machine itself, but it should have worked! You can clearly hear a hissing sound now when the valve diverts the airflow out of the tube where it simply blocked it before.

What I did:
[image]
The electronic valve on the left had to be flipped by 180 degress so that the tube coming from the compressor is now connected to the coupling that previously led into the tube towards the laser head and vice versa. The open coupling (facing the camera) now lets air escape into the machine's back when air assist is paused. This operation was made much more difficult than necessary by the second of the two screws holding the valve being nearly inaccessible, thankfully we had the flexible shaft screwdriver from the IFixIt toolkit (and a telescopic magnet for retrieving lost screws)...

The damaged coupling on the compressor can be changed whenever it arrives.

closed 2016-09-19 by LukasFreeze