heidelberg-makerspace/dai-makerspace-do-something #64: figure out recurring events in wiki calendar

LukasFreeze, 2016-09-16:
We can still not display recurring events correctly which makes the calendar look sadly empty. The current "process" (so far, some malfunctioning code from the MediaWiki wiki) is tracked in our wiki on the [events help page](https://wiki.heidelberg-makerspace.de/wiki/Help:Events#Recurring_Events).

chombium, 2016-09-19:
What happens when you try to create a recurrent event? Is there something shown?
Is the date format the right one as it is configured in MW?

Have you tried some of the [MW magic words (https://www.mediawiki.org/wiki/Help:Magic_words#Date_and_time)? Current timestamp {{CURRENTTIMESTAMP}} for example

```wiki
{{#set_recurring_event:Meeting
 |property=Has date
 |Has title=Making Hours
 |Has group=Recurring events          
 |start={{CURRENTTIMESTAMP}}
 |unit=week
 |period=1
 |limit=20
}}
```

If something is shown with the code above, than we should only find the right format for the date.

Best Regards

emka, 2016-09-20:
I do not remember exactly which variants we tried. Do you have a wiki account? You could play around yourself and see if some combination is working.

chombium, 2016-09-20:
I have an account, but I don't have write permission :-/
On the "view source" pages I don't see the preview button as well. Therefore I can not test.

Can someone please give me write permissions? My user is [Chombium](https://wiki.heidelberg-makerspace.de/wiki/User:Chombium)

LukasFreeze, 2016-09-20:
Done! Sorry, we have to block new users by default due to the amount of spammers we're seeing :-/

chombium, 2016-09-20:
Thanks @LukasFreeze! No worries, the spanners are annoying :)

chombium, 2016-09-26:
I've played a bit with the calendars and the recurrent events on my [user page](https://wiki.heidelberg-makerspace.de/wiki/User:Chombium) and it looks like that all of the recurrent events are created/read successfully, but they are not shown in the 'eventcalendar' not in 'calendar' format. 
It would be also interesting to see if there is something useful in  the web server logs, to see if php throws some errors somewhere, when rendering the recurrent events.

I've also checked the [eventcalendar page](https://www.semantic-mediawiki.org/wiki/Help:Eventcalendar_format) to see how it works. It is based on [fullcalendar jQuery plugin](https://fullcalendar.io/). The fullcalendar plugin does not recognize recurrent events and I don't see any errors in the javascript console. 

Therefore I think that there could be something wrong when the SMW PHP code tries to draw some events in the fullcalendar plugin.

Can someone check if there are any errors in the MW PHP logs?
Someone should simply try to (re)load my user page and check the logs to see if there are some errors

emka, 2016-09-26:
I visited your user page and checked the error log, but could not find any related error. In the table that is created on your page the entries do not have a date, maybe that is why they are not displayed in the calendar?

emka, 2016-09-27:
Something is actually broken, after your last edits the date appeared in the table and I find this in the logs:
```
[Tue Sep 27 07:05:32.579583 2016] [:error] [pid 2360] [client 66.249.76.83:50559] PHP Warning:  require_once(/var/lib/mediawiki-1.27.1/extensions/SemanticResultFormats/formats/ploticus/../../../maintenance/commandLine.inc): failed to open stream: No such file or directory in /var/lib/mediawiki-1.27.1/extensions/SemanticResultFormats/formats/ploticus/SRF_Ploticus_cleanCache.php on line 25
[Tue Sep 27 07:05:32.579709 2016] [:error] [pid 2360] [client 66.249.76.83:50559] PHP Fatal error:  require_once(): Failed opening required '/var/lib/mediawiki-1.27.1/extensions/SemanticResultFormats/formats/ploticus/../../../maintenance/commandLine.inc' (include_path='.:/usr/share/php:/usr/share/pear') in /var/lib/mediawiki-1.27.1/extensions/SemanticResultFormats/formats/ploticus/SRF_Ploticus_cleanCache.php on line 25
[Tue Sep 27 07:09:59.681300 2016] [:error] [pid 3308] [client 66.249.64.12:35255] PHP Fatal error:  Class 'SMWResultPrinter' not found in /var/lib/mediawiki-1.27.1/extensions/SemanticResultFormats/formats/ploticus/SRF_Ploticus.php on line 17
```

I should have fixed the broken path to ```commandLine.inc``` already, the missing ```SMWResultPrinter``` is not yet resolved.

chombium, 2016-09-27:
I've changed something and now the event shown are only on my user page...
I also found something about the ["No Result" problem](https://www.semantic-mediawiki.org/wiki/Help_talk:Eventcalendar_format#x.22No_Results.22_Yellow_errorbar_1547). 
In the answer some  datatypes declarations are mentioned. I think we should first check if the SMW is installed and configured correctly  and to solve the php error with the SMWResultPrinter.

I'll try to find some installation guide for SMW and some tests in order to check if everything is installed correctly.

The SMWResultPrinter error means that  the 'MW_INSTALL_PATH' environment variable is missing.
That's the code on [line 23](https://github.com/SemanticMediaWiki/SemanticResultFormats/blob/master/formats/ploticus/SRF_Ploticus_cleanCache.php#L23). checks if the variable is set and if it isn't it will set the include path relatively to the current file.

Check in the file (LocalSettings.php)[https://www.mediawiki.org/wiki/Manual:LocalSettings.php] it should contain a line
```
define('MW_INSTALL_PATH','/var/lib/mediawiki')
```
Adjust this line to the path where media wiki installation is located and the error should be fixed ;)

emka, 2016-09-27:
Your research was perfect, I added the ```MW_INSTALL_PATH``` and followed the "no result" hint to check if the properties are defined. We needed to set the type ```Date``` for [Property:Has_date](https://wiki.heidelberg-makerspace.de/wiki/?title=Property:Has_date&action=history) and now the calendar does show results!

chombium, 2016-09-27:
Good to read and see that it works now :) It was a bit hard to "debug" without ssh access to the server. Thanks for the assistance @emka. 

@LukasFreeze: Should we now check different recurrent event definitions and calendar filters before closing this issue?

LukasFreeze, 2016-09-27:
Wow, thanks for all your hard work @chombium! By the way, did you see there is a [Makerspace Twitter account](https://twitter.com/HD_Makerspace) now as you suggested a while ago? ;-)

I agree that we should leave this open for a bit longer, @emka will need some time to fully implement the changes anyway as far as I know.

chombium, 2016-09-28:
Ok. we leave it open until we check everything.

@LukasFreeze, @emka: maybe we can also create few labels for the issues in order to better see the status and the urgency of the issues. This issue would probably go under something like "solution proposed" and "needs verification". What do you guys think? Would such markings be useful?

Btw  Thanks for the link for the twitter account. Looks good, now we need to fill the timeline :)

LukasFreeze, 2016-10-06:
It works perfectly, I updated the events help page on the wiki and made "Weekly Making Hours" recurrent without any issues!

closed 2016-10-06 by LukasFreeze

chombium, 2016-10-07:
cool :-)