heidelberg-makerspace/dai-makerspace-do-something #66: commission table saw

LukasFreeze, 2016-09-20:
Our new table saw needs a bit of work before we'll start using it!

- [ ] make it lockable (nearly done)
- [ ] 3D print missing foot
- [ ] level the table (3 adjustable parts)
- [ ] find manuals, datasheets etc.
- [ ] document all findings on its [wiki page](https://wiki.heidelberg-makerspace.de/wiki/Table_Saw)

emka, 2016-09-21:
I designed and printed a foot and linked the original manual. The download link is only working sometimes, if it is not check out the Google chached version.

LukasFreeze, 2016-09-22:
Ciaran also leveled the table as far as possible without changing or adding parts, we'll have to see how well it works (not 100 % flat)

LukasFreeze, 2016-09-22:
I'll print the manual next Wednesday (could only get it through cached sites, but looks great!)

LukasFreeze, 2016-09-22:
I measured the provided vacuum adapter yesterday - instead of making an additional connector piece we should probably just replace the part entirely.

It is a square plate (approx. 53 x 53 mm) with 4 holes (40 mm apart) which I guess are 3 mm. If the screws are a bit bigger we can still widen these holes, but the distance is important of course. The part that needs redesigning is the tubular section in the center - now 30 mm OD, it needs to be smaller so the shop vac's tube (24 mm ID unstretched) can fit over it.

emka, 2016-09-22:
What is the thickness of the plate? And the difference of OD to ID? I uploaded my [current draft](http://www.thingiverse.com/thing:1785485) with 3mm plate, OD 24mm and ID 20mm.

LukasFreeze, 2016-09-22:
I didn't measure those to be honest, we'll have to see how much is enough to make the part stable. The thickness of the plate and the tube's walls looked to be about 3 mm in the original part so ID could be 18 mm.

emka, 2016-09-22:
Updated to ID 18mm. Another dimension I had to guess is the length of the pipe, currently 22mm.

LukasFreeze, 2016-09-22:
Sounds good, making it much longer will only reduce stability without any added functionality. Sorry about the flimsy measuring job...

emka, 2016-09-24:
The holes are actually 4mm, drilled them to fit. The OD could be a little larger for a tighter fit, or add additional rings to fit the vacuum adapter. But it works fine as it is.

LukasFreeze, 2016-09-29:
We have a problem with the fence, this issue is becoming larger than it used to be... It is only attached to the table in one spot and has quite some freedom to move/"bend" during the cut. This can cause the workpiece to get stuck on the blade and is potentially dangerous! I'm updating the initial post as well, details/suggestions here:

- [ ] replace the fence by _something_, e.g. an aluminium extrusion (40/40 could work, **double-check**), that is long enough to be clamped down at the opposite end
- [ ] add a second ruler at the opposite end (could be lasered with "MarkSolid" spray)
- [ ] (optional) ideally, the opposite end should not need a clamp but be fixable in a similar fashion as the front

Until then, only use the table saw with the fence clamped down at the opposite end or clamp down a **straight** piece of wood to use as a fence!

emka, 2016-10-12:
The saw blade is slightly skew (burning marks!), the "Spaltkeil" is too far on the right. We probably will reassemble the whole table when there is time.

Measuring by hand and clamping our long aluminum guide (wood has too much friction) is quite annoying, a second ruler and something more solid as a fence will be a major improvement to have fun using the table saw.

Another minor task:
- [ ] glue 3d printed foot to leg

LukasFreeze, 2016-10-13:
Andrea replaced the plastic thumbscrews on the fence by metal parts which greatly increased stability. She was also able to make the table more level in an area @beanobeano and I had struggled by increasing the size of one of the screw holes.
We then opened up the saw to fix the skewed blade and splitter (maybe that's the name?) and cleaned everything while we were at it ;-)

The safety hazards are taken care of now, I just finished cutting a few very accurate and unburnt pieces of plywood! I would still like to upgrade the fence (among a few other details) but that can go into an optional issue in the future. I'd say this one can be closed once there is a little more documentation on the wiki (and the foot is glued on)!

LukasFreeze, 2016-10-21:
further or optional steps in #84

closed 2016-10-21 by LukasFreeze