heidelberg-makerspace/dai-makerspace-do-something #68: improve wiki accessibility for new users

LukasFreeze, 2016-09-22:
The wiki is starting to be a great resource for several machines, but may be a little hard to use for unexperienced users. My suggestions to improve it (please add your own):

- [ ] write a "wiki how-to" page (already working on it)
- [ ] create a "guest account" to encourage people to write about their project - could be active on the Mac during Making Hours and password can be shared
- [ ] rewrite the note appearing on the "permission denied" page when new users try to edit a page but do not have write permission - @emka will have to see whether that is possible, but a bilingual version linking to the contact persons who can manage user rights would be great
- [ ] possibly create some incentive, e.g. a free drink for a project page?

emka, 2016-09-22:
To find out which text we need to edit, someone needs to open the page with "permission denied" error and add "&uselang=qqx" to the URL. This will show the name of the message we need to edit at "MediaWiki:$name".

emka, 2016-09-22:
We could use the Guest account without write permissions to do that.

LukasFreeze, 2016-09-22:
created the account _Maker_ with our usual easy password. Here's what comes up if you go to a page and try to edit it ("view source"):

> You do not have permission to edit this page, for the following reason:
> 
> The action you have requested is limited to users in the group: Write.

Here's the same paragraph with "&uselang=qqx":

> (permissionserrorstext-withaction: 1, (action-edit))
> 
> (badaccess-groups: (group-Write), 1)

In addition, there is this message when following a red link/trying to create a new page:

> There is currently no text in this page. You can search for this page title in other pages, or search the related logs, but you do not have permission to create this page.

Adding &uselang=qqx" to the URL does not have any effect in this case...

LukasFreeze, 2016-09-22:
Badaccess-groups works fine! I'd suggest we leave "Maker" without write permission until the next meetup in case further ideas to test come up.

LukasFreeze, 2016-09-26:
@Pakue95 suggested today that we could move (or copy) the "other media" links to the sidebar on the left.
While we are thinking about restructuring the main page anyway, it could be cool to have a "featured article" where the slideshow currently is and move the slideshow over to the blog.

LukasFreeze, 2016-11-14:
Started to write the "how to" guide at [Help:Wiki Basics](https://wiki.heidelberg-makerspace.de/wiki/Help:Wiki_Basics), any input is as always very welcome!

LukasFreeze, 2016-12-19:
tried to make the InfoBoxes a little more accessible by including a link to the documentation in the template today

LukasFreeze, 2017-02-08:
Started the project page ["Multi-Photo Imagemaps"](https://wiki.heidelberg-makerspace.de/wiki/Project:Multi-Photo_Imagemaps) to work towards more visual navigation