heidelberg-makerspace/dai-makerspace-do-something #71: establish proper printing/laminating section

LukasFreeze, 2016-09-26:
with the three new printers Luzian brought in the other week, we have quite a few to choose from but they have not been tested. We need to decide what to do with them this week!

- [ ] see what kinds of printers we have
- [ ] determine whether they work and could join the currently very lonely [laser printer](https://wiki.heidelberg-makerspace.de/wiki/Laser_Printer)
- [ ] build a storage solution for the ones we keep (e.g. shelves next to the windows)
- [ ] free up remaining ones for taking back home or disassembly - we can't have loads of unused equipment lying around...

and, a _little further_ down the road:
- [ ] commission laminator - find manual, get suitable laminating foil, test, add to the wiki, **ideally store with the printers**
- [ ] make the whole "2D printer farm" network accessible (could become a new issue, just to keep this in mind...)

Titan-OX, 2016-09-28:
We have:

- [Canon PIXUS iP4500](https://www.usa.canon.com/internet/portal/us/home/products/details/printers/support-inkjet-printer/ip-series/pixma-ip4500/pixma-ip4500) -A4 9600 x 2400 dpi
- [Canon PIXMA iP4300](https://www.usa.canon.com/internet/portal/us/home/products/details/printers/support-inkjet-printer/ip-series/pixma-ip4300) -A4 inkjet color printer, up to 9600x2400 dpi, can print photos

- [Epson Stylus Color 3000](https://www.epson.de/download.php?file=/files/brochures/c11c/c11c203-ger-sc3000-de.pdf&name=Epson-Stylus-Color-3000-Brochures-1.pdf) - A2

Pakue95, 2016-09-28:
Canon PIXUS iP4500 and Canon PIXMA iP4300 are very similar the ...4500 is just newer and has already some ink cartridges so we'll probably use this one if it works.

LukasFreeze, 2016-09-29:
Next step would be testing the iP4500; @Titan-OX says it should work but let's make sure before building the shelves. If all goes well, we'd need three shelves: iP4500, laser printer, laminator (just a small storage shelf, will be used on tables).
The iP4300 is in the junk cart for now, maybe somebody wants to disassemble it or have it for themselves (once we know the other one works!)

LukasFreeze, 2016-09-29:
**TL;DR: please don't build shelves yet!**

One more thing I forgot - there is a combination printer/scanner/copier sitting by the entrance (don't know who put it there, but by now I would assume it's meant for us...) that we should check as well. We've had a visitor say "who needs scanners with good smartphone cameras nowadays" but I personally would like having one. Thoughts?

JohannesGruhler, 2016-10-05:
driver download (Canon PIXUS iP4500) : http://www.canon.de/support/consumer_products/products/printers/inkjet/pixma_ip_series/pixma_ip4500.aspx?type=download&language=&os=WINDOWS%2010%20(64-bit)

LukasFreeze, 2016-10-06:
@JohannesGruhler had a closer look at the Pixus und the Epson printer/scanner combination someone left at the door and they both seem to work but are dry on all catridges. He will document the findings so far on the wiki page "Inkjet Printers".
As soon as that happens, I'll look into getting cheap new cartridges for at least one of them (will have to see if the Pixus is better in any aspect).

LukasFreeze, 2016-10-11:
Ink catridges arrived, we can test both A4 inkjets tomorrow!

emka, 2016-10-12:
I added my findings to the Laminator wiki page

JohannesGruhler, 2016-10-19:
Printer: Epson Stylus Photo PX700W works but the print 
result is not so good. see [https://wiki.heidelberg-makerspace.de/wiki/Inkjet_Printers](url)
Printer: Canon PIXMA iP4500 has problems with the (fake) ink cartridges

Pakue95, 2016-10-24:
Tested the laminator and it seemed to work with the 125µm foil. However the status LED is completely off and doesn't show the right temperature. Wait the recommended ~10min before laminating the first sheet instead of relying on the green status light, which turns on after 2-3 min.

LukasFreeze, 2016-11-01:
mounted one shelf today, two more are pretty much ready to go but will not fit if people don't like the new cpuch arrangement

JohannesGruhler, 2016-11-01:
Canon PIXMA iP4500:
I tried to put the ICs from the old ink containers to the new ink containers but the printer do not accept them too.

Pakue95, 2016-11-01:
It might be that the printer saves the ID of the container and with that the ink level. So using the same chip won't work. But you can get third party cartridges with a chip cheaply. 
Have a look at [these](https://www.amazon.de/Druckerpatronen-PGI-5BK-CLI-8BK-CLI-8C-schwarz/dp/B001V08294/ref=sr_1_2?s=computers&ie=UTF8&qid=1478043852&sr=1-2&keywords=CLI-8Bk).

LukasFreeze, 2016-11-02:
The ones we tested did have a supposedly compatible chip but were rejected. If the Epson works fine it may not be worth the money or time to keep working on the Canon...

LukasFreeze, 2016-12-07:
update on the A2 printer: Maik donated a US to EU adapter for both the plug and the delivered current! After some cleaning everything seemed to move alright and the place to change cartridges turned out to be the front right (push to open).

Printing could not be tested yet as proper software would be needed. If the machine turns out to work in the end, we will have to see where we can find a place for it...

LukasFreeze, 2017-01-05:
Jaro seems to have donated a color laser printer as well, no idea if it works right now.

JaroEiermann, 2017-01-06:
Hey Yeah, this was my christmas present. We tested it at Jaspers Home, it seems like ink is missing, but it is working in general.

LukasFreeze, 2017-01-06:
Cool, I'd like to try and connect one or two laptops to it next week but
then we can order some toner :-)