heidelberg-makerspace/dai-makerspace-do-something #73: Create email addresses

emka, 2016-10-04:
Create email addresses like ```orga@``` which forward mail to the responsible people's private mail account.

Pakue95, 2016-10-04:
While creating the `orgo@` mail forwards could you also create one dedicated for the twitter account? Right know it uses the Gmail one but this isn't ideal.

emka, 2016-10-11:
Done

LukasFreeze, 2016-10-12:
I'd like to keep this open as I think there is one thing left to do after creating these addresses - getting them out there! I put the YouTube contact onto the [YouTube wiki page](https://wiki.heidelberg-makerspace.de/wiki/YouTube), but the only place for FB/Twitter to go right now is the internal PR page.

The orga contact should be prominently featured, maybe in the sidebar?

emka, 2016-10-12:
I started the "Organisation" wiki page, should we add it to the sidebar already?

LukasFreeze, 2016-10-14:
I wouldn't mind, it will be appended in the near future but the important info is there :-)

emka, 2016-10-17:
Yes, I removed "Random page". If you liked it, I don't mind if you put it back.

The orga contact is now prominently featured in the wiki, I think this is done. @LukasFreeze if you like, you could update it on the DAI library page as well, but that is optional.

LukasFreeze, 2016-10-18:
I put it back in because I like to think it helps newbies discover the wiki if there are only a few minutes to spare. The sidebar (for non-administrators, of course...) is so empty otherwise ;-)

After some thinking I guess twitter and FB contacts can be internal, non-members don't usually have that much to share yet anyway, so let's mark this done. We can see about the contact on the DAI page when we update it anyway, but it probably looks better if there is a "dai" address - I'll forward the relevant stuff to orga.

closed 2016-10-18 by LukasFreeze