heidelberg-makerspace/dai-makerspace-do-something #74: 3D printer firmware should be managed in git

emka, 2016-10-05:
We have two different versions of Marlin on the Kossels and Prusas, and four different calibration settings.

- [ ] Find out which Marlin versions the manufacturer versions were forked from
- [ ] Clone the upstream Marlin and add branches for our Kossel and Prusa forks (starting from the historical commit)
- [ ] Create forks from the manufacturer branches to hold our individual printer calibration values
- [ ] Check out the latest upstream Marlin and merge with manufacturer modifications to have current firmware on all printers with individual calibration settings

LukasFreeze, 2016-12-07:
This is not something I as a git noob will try on my own, but once we have accomplished this I suggest the first change to all firmwares to be a check whether the G-code was sliced with the correct profile in Cura (where we would add an appropriate comment line in the beginning, e.g. "Heidelberg Makerspace i3 A"

emka, 2016-12-11:
I was trying to find out which commit a changed source tree was based on and wrote a little quick-and-dirty script, which could be reused for this purpose:

```bash
#!/bin/bash

NUM_COMMITS=100
COMPARE_DIR=../other_source/

# get list of last commits
commitlist=$(git log --oneline | head -n$NUM_COMMITS | awk '{print $1}')
original_commit=$(echo $commitlist  | awk '{print $1}')

for commit in $commitlist
do
	git checkout -q $commit

	# count differences
	num_differences=$(diff . $COMPARE_DIR | wc -l)
	echo $commit $num_differences
done

git checkout $original_commit
```

emka, 2016-12-11:
@LukasFreeze I could run that script for you if you give me the manufacturer's source code for both printer types. Not sure what @DirkToewe found out already.

LukasFreeze, 2016-12-14:
the Kossels run on "Rich Cattel Marlin" which was [last updated in late 2015](https://github.com/RichCattell/Marlin) - close enough, we built them in April that year

emka, 2016-12-15:
In that case I think we could just switch to [upstream Marlin](https://github.com/MarlinFirmware/Marlin). If I got it correctly the "RIchCattel" fork was created to add delta printer support, but that is supported in upstream now as well. The current release is 1.0.2-2, maybe we should wait for 1.1 (RC8 has known issues that I don't like) or just start adapting the default configuration for 1.0.2 already.

emka, 2016-12-15:
I forked the upstream project and created [a Kossel branch](https://github.com/heidelberg-makerspace/Marlin/tree/Makerspace-Kossel) from the ```1.0.2-2``` tag.