heidelberg-makerspace/dai-makerspace-do-something #76: milling table needs power and light

LukasFreeze, 2016-10-09:
The milling motor arrived and we should be able to power it via the extension cord I laid there from the 3D printer farm, but that's not ideal and it is also pretty dark in that corner.

My idea would be to use the power outlet in the ceiling strip connected to a triple extendion cord mounted on the wall, and some kind of lamp with a plug. We should mount a board to the wall rather than drilling holes into the wall for every single part of this solution.

Any other suggestions?

emka, 2016-10-10:
Ceiling strip sounds good, we should ask Andrea for a professional opinion. If that does not work, there is also a power outlet left of the green workbench, which is the closest one. A cable could be easily mounted below the shelf.

LukasFreeze, 2016-11-03:
Mounted a large board to the wall yesterday, attaching a power strip and lamp is easy now. The lamp that was included in André's original donation might still work, but the bulb is definitely broken (E14, 40 W) and it needs an extension to reach beyond the top of the table which would cast a big shade if the lamp is right on the wall.

LukasFreeze, 2016-11-24:
Donated a suitable LED bulb and power strip, both are waiting to be mounted. For the power strip, a 3D printed holder like the one @emka used in the Stepcraft room would be nice! The light fixture works but has a hidden switch that may make it look otherwise - needs to be marked ;-)

A solid beam would be alright for the light at first but we should really make something cool and adjustable eventually!

emka, 2016-11-24:
Here are the holders: https://www.thingiverse.com/thing:209651

LukasFreeze, 2016-11-26:
The power strip is mounted (nails for the cord, double sided tape and printed brackets for the outlets) and I started a lamp mount.

LukasFreeze, 2016-11-28:
Extendable lamp arm done, wall mount hopefully doable on Wednesday. 

Also noticed something today: as a side effect of taking power from the ceiling strip the milling table can only operate with the lights on, but I think that just makes it safer ;-)