heidelberg-makerspace/dai-makerspace-do-something #79: Set up donated computer

emka, 2016-10-12:
Andre donated an old computer, we need to 
- [ ] make space for it on a table,
- [ ] find monitor, keyboard, mouse (near printer farm?) and
- [ ] install a current operating system (I suggest Ubuntu 16.04 LTS).

emka, 2016-10-19:
We might need a wifi stick for internet access. The Freifunk router near the window should ideally get cable access, then we could have Ethernet cables from the router to Mac mini and the donated computer.

emka, 2017-01-08:
This is a Pentium 4 (32bit). Will install Ubuntu 16.04.1 as soon as the image is downloaded.

emka, 2017-01-08:
It did not accept my boot stick ("NTLDR fehlt"). The computer will probably be too slow to do anything with it other than reading 3.5" floppy disks.

LukasFreeze, 2017-01-08:
Too bad, in that case we wouldn't need to keep it - we have a few external
floppy drives (under the workbench I believe) in case anyone will ever need
to read old disks.

emka, 2017-01-26:
Maybe we are still able to install Ubuntu, lets see. At least we got a USB wifi stick now.

ragflyer, 2017-01-27:
I am running Ubuntub16 on one old netbook and Xubuntu16 on the other one. Might be the low memory of 1gb but the desktop of Ubuntu is unusable while xubuntu is quite responsive.