heidelberg-makerspace/dai-makerspace-do-something #82: Pfaff creative 1.5 bobbin spooler doesn't turn

LukasFreeze, 2016-10-18:
The bobbin spooler (little pin on the right at the top of the machine) of the Pfaff creative 1.5 sewing machine does not turn under workload and only veeeery slow without a bobbin on it.

Opening the top cover is not trivial so we will talk to the people at the store before voiding warranty!

LukasFreeze, 2016-10-25:
Apparently the spooling did use to work at some point (at least when the machine was brand new), will go to the store soon if I have time...

LukasFreeze, 2016-11-08:
There were more problems, e.g. somebody apparently opened the machine's "Stichplatte" and let a screw fall into the mechanics! That's great...machine is on its way to the manufacturer.

jasperclaudio, 2016-12-14:
Die Maschine ist wieder da mit einer erweiterten Garantie auf Ersatzteile! Bitte sorgsam mit ihr umgehen und am besten nur in Anwesenheit eines*r erfahren*en Benutzer*in verwenden... es ist jetzt schon das zweite Mal dass wir sie in die Werkstatt bringen mussten.

closed 2016-12-14 by jasperclaudio