heidelberg-makerspace/dai-makerspace-do-something #84: improve table saw

LukasFreeze, 2016-10-21:
The [table saw](https://wiki.heidelberg-makerspace.de/wiki/Table_Saw) is usable now, so #66 is closed and optional steps from there (and others) are coordinated here to further improve its performance and safety.

- [ ] glue 3D printed foot to leg (currently piece of duct tape)
- [ ] attach pieces of anti-slide mat to the feet
- [ ] get blade change tools from previous owner
- [ ] replace all PH screws with internal hex screws for less cam-out (two cylindrical heads for the ends of the sled grooves, eight countersunk screws for the red plastic surrounding the blade - all *probably* M4 x 8 mm)
- [ ] improve the parallel fence - ideally longer profile to reach the opposite side of the table, second clamp and ruler in the back

LukasFreeze, 2016-11-17:
all 10 screws are now M4 x 10 mm countersunks with internal hex drive, the matching key is magnetically attached to the left front edge of the undercarriage!

LukasFreeze, 2016-11-17:
Live now: [Project:French Cleat Storage](https://wiki.heidelberg-makerspace.de/wiki/Project:French_Cleat_Storage)! Lots of homeless tool should go on that wall soon :-)

LukasFreeze, 2016-11-24:
Any news on the blade change tools? I saw a hack/workaround online that could work in case we are not able to get a hold of the previous owner...

emka, 2017-04-27:
I have met the previous owner again a while ago. He could not find the tools anymore and promised to contact me if he finds them. I think we should better find a replacement.

LukasFreeze, 2017-04-27:
They should be relatively standard and can be bought along with a new blade
once needed - thanks for checking anyway!