heidelberg-makerspace/dai-makerspace-do-something #85: Permission denied when saving wiki

emka, 2016-10-23:
Mobile view and translations just show "permission denied" when saving.

The failing request is a ```POST``` to ```api.php``` with ```action=edit```, response is ```{"error":{"code":"permissiondenied","info":"Permission denied","*":"See https://wiki.heidelberg-makerspace.de/mediawiki/api.php for API usage"}}```.

emka, 2016-10-23:
According to https://www.mediawiki.org/wiki/Extension_talk:VisualEditor the error is caused by the Lockdown extension we are using for internal pages.

@LukasFreeze do you think we could get rid of internal pages? Then we could also sync to Github like RZL does at https://github.com/raumzeitlabor/wiki/
Aside from the Internal: namespace, the Begeisterhaus: namespace is also using the Lockdown extension, but there is not content yet in this namespace.

LukasFreeze, 2016-10-23:
The internal pages are not really sensitive information - with the exception of "Funds" which I would delete then. Maybe having a namespace called for these things that the public does not really _need to_ care about would still be nice, just no special permissions?

As for the begeisterhaus stuff, I'd suggest we ask the other two or three via that mailing list first, but I doubt much is going to be added there anytime soon...

emka, 2016-10-23:
The talk page I linked may be the wrong one, but disabling Lockdown resolved this problem when I tested it. Keeping it enabled until we decided how to proceed.

emka, 2016-10-24:
The actual issue is https://phabricator.wikimedia.org/T148582

As a temporary solution we will remove internal pages and disable Lockdown.

emka, 2016-10-24:
Deleted sensible internal content, disabled Lockdown, mobile editor and marking edits as patrolled is working again.

closed 2016-10-24 by emka